brycesub/silvia-pi #2: Document installation instructions for brew PID control

brycesub, 2016-01-22:


brycesub, 2016-03-01:
Full instructions aren't up yet, but pictures documenting disassembly and mounting are available here: http://imgur.com/a/3WLVt

jjeaton, 2016-03-17:
@brycesub very interested in trying out this project on my Silvia, any timeline on when you'll write the instructions? I'm having a hard time following the pictures.

brycesub, 2016-03-17:
I'm happy to give you a walkthrough via skype.  Do you have parts ordered?

jjeaton, 2016-03-22:
@brycesub that is very generous of you. I haven't ordered parts yet, primarily because I was trying to figure out whether I had the skills to complete the project.

The two things that are drawing me to this are:

* Control the machine via wifi
* Potentially cheaper option than buying a PID (which doesn't support controlling over wifi)

I am a programmer, but have limited electronics experience (I built some effects pedals for my guitar about 10 years ago), and I'm terrible at soldering. So I was hoping to see what the instructions would entail to know what other supplies I may need to determine whether I could do it and whether it would be cheaper than buying a PID, if I was willing to give up the wifi aspect.

Can this control steam temperature or does it not yet do that?
Do you have any idea what the differences would be if you used the Raspberry Pi Zero instead of the board you used? (cost reasons)

I may take you up on the Skype, but it wouldn't be for awhile. Super cool project, thanks for sharing!

brycesub, 2016-03-22:
@jjeaton The pi-zero should work just fine for this project assuming you got the micro-usb to usb-A adapter to connect the Wi-Fi adapter.  Additional supplies needed would be a soldering iron and some solder.  You only actually need that to solder the two wires from the thermocouple to the thermocouple-breakout.  Everything else is connected via jumper or just crimped w/ pliers.

It doesn't do steam temp control (yet), I just didn't really see a need for precision on the steam sided (compared to the coffee side).  I have ambitions to add another relay and controlling the pump later down the road.

I'm not an electrical engineer by trade, I'm a database administrator with minimal electronics experience.  I know you can do this.

jjeaton, 2016-04-01:
@brycesub I appreciate your show of faith! :) I assume you'd rather have a conversation after I have the parts. Will come back once I do. Probably will go with the Pi 2, not the zero.

brycesub, 2016-04-01:
@jjeaton Grab a pi 3, then no need for the wifi adapter.

obiben, 2016-09-18:
So @jjeaton did you go through with it?

Thought I'd write a word here since I've just ordered parts to do something similar. I'll most likely fork this when I receive the pi (zero) and screen. Planning on doing this offline with a touchscreen embedded in the machine where the rancilio logo is.

More news in the coming weeks!

jjeaton, 2016-09-19:
I almost pulled the trigger and ended up getting a good deal on a Breville dual boiler and sold my Silvia. The hacker in me is sad, but the coffee is happy. 😄

greymda, 2017-01-09:
can't figure it out from the pictures where did you mount the thermocouple?

brycesub, 2017-01-09:
I mounted it where the old brew thermostat used to be, reusing one of the screws and screw hole.

brycesub, 2017-02-12:
See https://github.com/brycesub/silvia-pi/issues/10 for information for wiring the breakout and relay to the pi.

obiben, 2018-01-10:
Well, it took a lot of spare time to figure out how to make my pitft screen without breaking wifi on Pi 0 W, but it ended up working. 

[image]

therealjohn, 2018-01-19:
@obiben Now that's a mod! Did you talk about what you've done to the Silvia? Thanks for sharing!

obiben, 2018-01-19:
I'm in the process of documenting it on my [fork](https://github.com/obiben/silvia-pi)
Need to get the CAD files for the plexi out of my old laptop sometime soon