brycesub/silvia-pi #9: Gaggia Classic installation?

greymda, 2016-10-20:
i wonder... can it be done on a Gaggia Classic?

brycesub, 2016-10-21:
Indeed it can, the internal wiring would obviously be slightly different (but the wiring in both machines are very similar).  In fact I just purchased an old Gaggia Classic on ebay to refurbish and I'll be putting one in it when I'm done.  If you want to give it a go I would make myself available to assist on getting it wired up.

greymda, 2016-10-24:
that would be great! but mine is on its way to me (eBay, too) and i need to see its condition when i get it. will definitely reach you when i have all the items. if you start earlier, please make some notes! kind regards and thanks for your help.

dliuzzi, 2017-04-26:
I'll also interested in this. I own a pre-2015 Gaggia Classic and I've been looking to adding a PID to it. I'm more inclined to this projects than those premade kits they sell as it seems way customizable. If I can help with anything I'll be glad to!