brycesub/silvia-pi #10: MAX31855K to GPIO

sebastian-alfers, 2017-02-10:
Hello,
thanks for sharing your project.

Could go define how to connect the breakout to the GPIO? Not sure how to connect them proper.

Thanks

brycesub, 2017-02-12:
Use these two pictures as a guide:
http://i.imgur.com/cALY8cP.jpg
http://i.imgur.com/niWjavD.jpg

Pinout for pi2 and pi3:
https://az835927.vo.msecnd.net/sites/iot/Resources/images/PinMappings/RP2_Pinout.png

From the thermocouple breakout board connect:
GND -> Any ground pin (I used Pin 20)
VCC -> 3.3v Power (Pin 17)
SCK -> SPIO SCLK (Pin 23)
SO -> SPIO MISO (Pin 21)
NC -> No connection
CS -> SPI0 CS0 (Pin 24)

Hope this helps!  Let me know if you need any other info.

closed 2017-02-12 by brycesub

sebastian-alfers, 2017-03-08:
@brycesub Thank you for your hints regarding GPIO. Got the thermostat running!

Sorry, just another quick question. Which output of the relay (1 and 2) needs to go to B1 and B2 in this picture http://i.imgur.com/gNaIdi8.jpg?
Thanks!

brycesub, 2017-03-08:
It actually doesn't matter, you can hook it up either way.