brycesub/silvia-pi #15: Thermocouple question

jfbibeau, 2017-08-24:
Hey there - currently buying parts for the project, can you clarify if you bought the thermocouple with the mini connector or the regular spade; and if so what did you have to do to solder to the breakout board? I can't see the connection between thermocouple and breakout and the pictures... I have no issue with soldering, but was it easy to split the two wires from the mini connector?

Thanks!

brycesub, 2017-08-24:
The mini connector unscrews easily and the thermocouple wires are held inside the connector by more screws.  Once the connector is removed the two wires are soldered to the breakout board.

jfbibeau, 2017-08-24:
Awesome thanks for the quick reply!

closed 2017-08-24 by jfbibeau

jfbibeau, 2017-08-31:
@brycesub Hi again! One more question, what type of quick connect/crimp (male?) connectors did you put at the end of the 14 gauge wires formerly connected to the brew thermostat? Any special type to make sure they fit with the existing female connectors of the Silvia? 

Something like this? https://www.fastenal.com/products/details/58569

Thanks!

brycesub, 2017-08-31:
The link you posted isn't working (for me at the moment, might be maintenance on their side).  I got some spade terminals (at Home Depot, I think), and then used some pliers to crimp them into the existing connectors on the wires that go to the heating element.  I dug out the package, here it is:
http://i.imgur.com/QeAlLGY.jpg  AMP Spade Terminals 16-14.  Good luck!

jfbibeau, 2017-09-04:
Thanks for the help, successful install!

[image]