brycesub/silvia-pi #18: Thermocouple temp drops when Silvia is plugged in

jfbibeau, 2017-09-05:
Hey again!

Did you ever have any issues where the thermocouple temperature readings drop when you plug the Silvia in?

I could've sworn I had the whole thing working, then when I re-assembled the machine and tried it, the boiler was way too hot and pushing steam out the group head. Took everything apart again, and noticed that with the machine disconnected the idle temp is ~80F, but as soon as I plug in the machine (don't even flip the power switch), the temp readouts drops to 60F... so all my temps are 20F off which explains the steam.

Have you encountered that at all? It looks like AC interference or a grounding problem... :(

jfbibeau, 2017-09-05:
Looks like the MAX breakout board can't handle grounded thermocouples, and the boiler on Silvia is grounded (so is the thermocouple)...

I'm either getting the nonsensical temp above (~20F off), or `nan` with the fault code of `shortGND`. 

Did you run into this at all? Thanks in advance..

brycesub, 2017-09-05:
Hmm...  I have not seen this issue with my setup.

Which breakout board are you using?  This is the one I used: https://www.sparkfun.com/products/13266

Can you send a picture of how you have the thermocouple mounted?

jfbibeau, 2017-09-05:
Yep, I'm using the exact same breakout board and thermocouple as you.

I can send you a pic of the thermocouple mount tonight, it looks similar to yours. I just screwed it next to the old brew thermostat.

But it doesn't appear to be a mount issue. Basically I've got it unmounted right now, and if I hold it in my hand, the temp shows ambient air temp ~80F, but as soon as I tap it against the boiler with my hand, the graph temp drops to ~60F. Then if I remove it, the temp goes back to 80F, almost like if the boiler was 20F cooler (it isn't, confirmed it with a separate temp probe haha).

It really looks like involving the thermocouple in the Silvia's grounding circuit is messing with the voltage read from it. I'm really surprised you haven't seen this - I mean you may not have measured temp from the probe on/off the boiler to compare, but I imagine you also don't have crazy steam coming out of your grouphead so presumable it's reporting the right temp in your case.

I could put a ~20F compensation in the PID loop but that sounds crappy :(

brycesub, 2017-09-05:
I'll take another peek at mine tonight, but the ambient temperature of the room and the reading of the boiler report accurately when the whole machine is at rest (and plugged in).  What version of the Silvia do you have?  Is it possible my boiler is ungrounded?

jfbibeau, 2017-09-05:
Looks identical to yours, I believe it's a v3? 

I can tell from your picture here: http://i.imgur.com/YMIVbTj.jpg that your boiler is grounded (see the hex screw on the left with 2 green connectors to it). 

Very strange... I don't know what else it could be. How are you powering your raspberry PI? Is it a 2 prong USB -> wall adapter?

jfbibeau, 2017-09-06:
[image]

Thermocouple mount picture.

jfbibeau, 2017-09-06:
What happens when I unplug the machine from the wall :(

[image]

brycesub, 2017-09-06:
I tried the same thing last night, viewed the UI, and then unplugged the espresso machine AC from the wall and didn't notice any noise at all in the temperature, stayed steady at ambient (~76 F).  Can you double-check that the ground from the breakout board is connected to a ground pin on the pi, maybe try a different ground pin?  Also, is the NC pin not connected to anything?

brycesub, 2017-09-06:
To answer your previous question, I am using a 2 prong USB adapter to power the pi.

jfbibeau, 2017-09-06:
Yeah the ground pin from the breakout board is connected to a ground pin on the pi. Tried different ground pins, same effect...

The NC pin from breakout board is not connected to anything - I didn't solder anything to it at all (I noticed on your picture you have a cable connected to it, but can't tell what you connected it to?)

Did you put any thermal paste or anything between the thermocouple and the brewer?

This is really bugging me haha. Unless you ended up with a different thermocouple than me... Is your thermocouple sleeve made out of stainless steel?

brycesub, 2017-09-06:
I soldered on a pin for the NC pin on the breakout board, but it isn't connected to the pi (or anything).

I'm unsure about what the sleeve of the thermocouple is made of, though it appears to be SS.

I did re-use some of the thermal paste that was between the old thermostat and the boiler... perhaps I got extremely lucky.  Let me check tonight and report back.  There should be some material you can use that would conduct heat but not electricity as a buffer between the boiler and thermocouple.

jfbibeau, 2017-09-06:
Yeah I have a tube of thermal paste I could use... but I'd have to get extremely lucky for none of the probe or sleeve to end up touching the boiler (or the screw that runs through the washer) at all :)

I did an extremely crude test last night and wired up my un-grounded Fluke thermocouple to the breakout board and to the boiler, and did not see temperature fluctuations... I might try to order an ungrounded thermocouple to see if it fixes it, if I can find one where I won't get dinged with so much shipping and duty :)

brycesub, 2017-09-06:
Did you order your original thermocouple from the link in the README.md?

jfbibeau, 2017-09-06:
Yes, same link as README from auberins

jfbibeau, 2017-09-13:
Bought and received a new thermocouple with nylon sleeve (ungrounded), and same issue.

*hits head on desk*

brycesub, 2017-10-03:
Have we eliminated your Pi as a potential source of the problem?  What model are you using (I have a pi2 in my machine).

jfbibeau, 2017-10-04:
I haven't ruled that out yet. I have a model B. I have a 2nd one lying
around I could try. I contacted Sparkfun to see if it was a defect with the
breakout board, they refunded me... I bought a new one from Amazon and same
issue. So ruled out thermocouple and breakout board... That just leaves the
espresso machine or the Pi. I'm getting close to giving up :)

jfbibeau, 2017-10-12:
Switched Pi - no difference. 

However I kind of found one solution which works: I used the non-grounded thermocouple I bought, held it in place by placing it under the now un-used brew thermostat and screwing it back in place, then covered the thermacouple probe with thermal compound. That way the thermocouple probe is not grounded, but the temperature reported seems pretty accurate. I no longer have the temp drop when plugging in the machine. 

I can't help but come to the conclusion that the only way this will ever work with the MAX31855 is if the probe does not actually ever touch the boiler, so thermal compound is a must to electrically isolate it.

brycesub, 2017-10-12:
I can't believe that I was that lucky with the placement of my thermocouple.  I'll look to see if there is a more appropriate thermocouple amp.  Have you come across any?

jfbibeau, 2017-10-12:
Yeah, I find it really hard to believe too. That being said, my first wiring attempt (with everything external), everything was working fine also - only when taking it apart and re-wiring it inside the Silvia did things start to go wrong, so I was lucky on my first attempt too? I don't know, seems strange, but I've ruled everything out :(

I haven't done much research on other thermocouple amps. I got a refund on the Sparkfun one because they couldn't figure out what the problem was - I'm currently using this one: https://www.amazon.ca/gp/product/B073HZFF2W/ref=oh_aui_detailpage_o00_s00?ie=UTF8&psc=1 but it's still based off the MAX31855, so essentially the same thing.

This guy seems to be an upgrade and works with other thermocouple types: https://www.adafruit.com/product/3263 but might have the same problem.

Another way I've seen people do it is use mica washers + teflon tape around the screw, and that would isolate it.

jfbibeau, 2017-10-21:
Alright. It's finally working. Ungrounded thermocouple that doesn't touch the boiler, with thermal paste. We're finally in business and the machine is reassembled and reporting accurate temps! Thanks for all your help! :)

closed 2017-10-21 by jfbibeau

obiben, 2018-01-10:
@jfbibeau Did you happen to notice if there was variation within the sensor's state? When I print out my sensor.readState(), shortGND and shortVCC come True once in a while, not sure if they follow a pattern but they don't seem to be True at the same time.

jfbibeau, 2018-01-10:
Yes, I had the exact same behavior you described.

obiben, 2018-01-10:
Hmm dang, have no way of seeing whether I get temp changes when I ground the thing since I've wired my pi directly to the silvia's power switch. Tried pinching the probe under the original thermostat with thermal compound I had left from my PC under it but to no avail.

jfbibeau, 2018-01-10:
Does your probe report temperatures correctly if it doesn't touch the boiler at all? You can pinch it in your fingers or blow on it and you should see corresponding temperature rise/fall in the graph.

obiben, 2018-01-10:
Yup. I've done lots of testing outside the machine. It seems to be working despite the error codes as well, except for the fact that the temperature seems to be hovering 1 degree under target. 

What made me wonder is this morning (first day of usage) it started working fine for my girlfriend, but when I got up half an hour later, I had a perfectly flat line at 95C. This was caused by my modification to eliminate outliers (during testing I'd get values of -20C once in a while). Whenever I get an outlier I use avgtemp as curtemp. I SSH'd and tail'd my process and it was reporting 0C all over with ground faults which brought me here.

I've ordered thermal conductive tape, which should be more reliable to isolate than grease. I'll update tomorrow or the day after with results.

jfbibeau, 2018-01-10:
Yeah, that should fix it. Let me know what your experience with thermal conductive tape is! I considered it but since I had thermal paste around I just decided to use that.

Fingers crossed its been solid for a few months now :D

(I think 1 degree under target is pretty reasonable, dont think K type thermocouples are that accurate. From what I read an error margin of +/- 2C is normal. Plus you can adjust the desired target temp through the UI so not too hard to account for it!)

obiben, 2018-01-10:
On a side note, pinching the probe with the thermostat affords me the opportunity to see what happens with the stock configuration! The thermostat clicks off at 85C, then temp rises to 92, drops to a whopping 75 before it clicks on again. At 94 I start seeing pressure rising from some steam+air expanding which tell me the probe isn't far off.

jfbibeau, 2018-01-10:
Yeah I remember doing the same experiment, the temperature curve is pretty wild on the stock thermostat!

I really don't miss having to temperature surf with this PID...

obiben, 2018-01-12:
Hmm.. seems like it might be a different issue. With the tape, still did the same thing. For shits and giggles, I took the probe out from under the thermostat and ran this code:

```
import Adafruit_MAX31855.MAX31855 as MAX31855
import time

sensor = MAX31855.MAX31855(13, 26, 19)

while True:
  print sensor.readState(), sensor.readTempC(), time.time()
```

Below is what my output is (tldr: goes from OK to shortGND to shortVCC in no particular order).

The sensor is this one: https://www.amazon.ca/gp/product/B073HZFF2W/ref=oh_aui_detailpage_o03_s00?ie=UTF8&psc=1

Now that I'm paying attention, I guess I should have looked at the only review. I was only looking for a sensor with the good circuitry that shipped free in 2 days since I had fried the original (a sparkfun one). 

Will code a workaround and order a new one!

```
{'fault': True, 'openCircuit': False, 'shortVCC': True, 'shortGND': False} nan 1515715831.56
{'fault': True, 'openCircuit': False, 'shortVCC': True, 'shortGND': False} nan 1515715831.56
{'fault': True, 'openCircuit': False, 'shortVCC': True, 'shortGND': False} nan 1515715831.57
{'fault': False, 'openCircuit': False, 'shortVCC': False, 'shortGND': False} 44.75 1515715831.58
{'fault': False, 'openCircuit': False, 'shortVCC': False, 'shortGND': False} 44.75 1515715831.59
{'fault': False, 'openCircuit': False, 'shortVCC': False, 'shortGND': False} 44.75 1515715831.59
{'fault': False, 'openCircuit': False, 'shortVCC': False, 'shortGND': False} 44.75 1515715831.6
{'fault': False, 'openCircuit': False, 'shortVCC': False, 'shortGND': False} 44.75 1515715831.61
{'fault': False, 'openCircuit': False, 'shortVCC': False, 'shortGND': False} 44.75 1515715831.62
{'fault': False, 'openCircuit': False, 'shortVCC': False, 'shortGND': False} 44.75 1515715831.63
{'fault': False, 'openCircuit': False, 'shortVCC': False, 'shortGND': False} 44.75 1515715831.63
{'fault': False, 'openCircuit': False, 'shortVCC': False, 'shortGND': False} 44.0 1515715831.64
{'fault': False, 'openCircuit': False, 'shortVCC': False, 'shortGND': False} nan 1515715831.65
{'fault': True, 'openCircuit': False, 'shortVCC': True, 'shortGND': False} nan 1515715831.66
{'fault': True, 'openCircuit': False, 'shortVCC': True, 'shortGND': False} nan 1515715831.67
{'fault': True, 'openCircuit': False, 'shortVCC': True, 'shortGND': False} nan 1515715831.67
{'fault': True, 'openCircuit': False, 'shortVCC': True, 'shortGND': False} nan 1515715831.68
{'fault': True, 'openCircuit': False, 'shortVCC': True, 'shortGND': False} nan 1515715831.69
{'fault': True, 'openCircuit': False, 'shortVCC': True, 'shortGND': False} nan 1515715831.7
{'fault': True, 'openCircuit': False, 'shortVCC': True, 'shortGND': False} nan 1515715831.71
{'fault': True, 'openCircuit': False, 'shortVCC': True, 'shortGND': False} nan 1515715831.71
{'fault': True, 'openCircuit': False, 'shortVCC': True, 'shortGND': False} nan 1515715831.72
{'fault': False, 'openCircuit': False, 'shortVCC': False, 'shortGND': False} 43.75 1515715831.73
{'fault': False, 'openCircuit': False, 'shortVCC': False, 'shortGND': False} 43.75 1515715831.74
{'fault': False, 'openCircuit': False, 'shortVCC': False, 'shortGND': False} 43.75 1515715831.75
{'fault': False, 'openCircuit': False, 'shortVCC': False, 'shortGND': False} 43.75 1515715831.75
{'fault': False, 'openCircuit': False, 'shortVCC': False, 'shortGND': False} 43.75 1515715831.76
{'fault': False, 'openCircuit': False, 'shortVCC': False, 'shortGND': False} 43.75 1515715831.77
{'fault': False, 'openCircuit': False, 'shortVCC': False, 'shortGND': False} 43.75 1515715831.78
{'fault': False, 'openCircuit': False, 'shortVCC': False, 'shortGND': False} 43.75 1515715831.78
{'fault': False, 'openCircuit': False, 'shortVCC': False, 'shortGND': False} 44.0 1515715831.79
{'fault': False, 'openCircuit': False, 'shortVCC': False, 'shortGND': False} nan 1515715831.8
{'fault': True, 'openCircuit': False, 'shortVCC': True, 'shortGND': False} nan 1515715831.81
{'fault': True, 'openCircuit': False, 'shortVCC': True, 'shortGND': False} nan 1515715831.82
{'fault': True, 'openCircuit': False, 'shortVCC': True, 'shortGND': False} nan 1515715831.82
{'fault': True, 'openCircuit': False, 'shortVCC': True, 'shortGND': False} nan 1515715831.83
{'fault': True, 'openCircuit': False, 'shortVCC': True, 'shortGND': False} nan 1515715831.84
{'fault': True, 'openCircuit': False, 'shortVCC': True, 'shortGND': False} nan 1515715831.85
{'fault': True, 'openCircuit': False, 'shortVCC': True, 'shortGND': False} nan 1515715831.85
{'fault': True, 'openCircuit': False, 'shortVCC': True, 'shortGND': False} nan 1515715831.86
{'fault': True, 'openCircuit': False, 'shortVCC': True, 'shortGND': False} nan 1515715831.87
{'fault': False, 'openCircuit': False, 'shortVCC': False, 'shortGND': False} 43.75 1515715831.88
{'fault': False, 'openCircuit': False, 'shortVCC': False, 'shortGND': False} 43.75 1515715831.89
{'fault': False, 'openCircuit': False, 'shortVCC': False, 'shortGND': False} 43.75 1515715831.89
{'fault': False, 'openCircuit': False, 'shortVCC': False, 'shortGND': False} 43.75 1515715831.9
{'fault': False, 'openCircuit': False, 'shortVCC': False, 'shortGND': False} 43.75 1515715831.91
{'fault': False, 'openCircuit': False, 'shortVCC': False, 'shortGND': False} 43.75 1515715831.92
{'fault': False, 'openCircuit': False, 'shortVCC': False, 'shortGND': False} 43.75 1515715831.93
{'fault': False, 'openCircuit': False, 'shortVCC': False, 'shortGND': False} 43.75 1515715831.93
[... cut ...]
{'fault': True, 'openCircuit': False, 'shortVCC': True, 'shortGND': False} nan 1515715832.8
{'fault': True, 'openCircuit': False, 'shortVCC': True, 'shortGND': False} nan 1515715832.81
{'fault': True, 'openCircuit': False, 'shortVCC': True, 'shortGND': False} nan 1515715832.82
{'fault': True, 'openCircuit': False, 'shortVCC': True, 'shortGND': False} nan 1515715832.82
{'fault': True, 'openCircuit': False, 'shortVCC': True, 'shortGND': False} nan 1515715832.83
{'fault': True, 'openCircuit': False, 'shortVCC': True, 'shortGND': False} 44.0 1515715832.84
```

the-dbp, 2020-04-09:
I don't understand why this issue was closed. the project should be updated to avoid grounding the thermocouple. either by adding a material to insulate the thermocouple or using an ungrounded thermocouple

on adafruits official website:

> The MAX31855 does not support grounded thermocouples - if the sensor touches ground the chip will return an error

i understand that the project isn't new anymore, but it's odd to me that it ever worked.

TiStef, 2020-04-28:
> Alright. It's finally working. Ungrounded thermocouple that doesn't touch the boiler, with thermal paste. We're finally in business and the machine is reassembled and reporting accurate temps! Thanks for all your help! :)

good morning, if and when you have time would you mind sending pics of your installation and link to the ungrounded wire you purchased? many thanks!

jfbibeau, 2020-04-28:
@TiStef Sorry I just looked and somehow didn't end up taking a picture of the final setup :( But basically imagine the thermocouple resting slightly above the boiler, with a thin layer of CPU thermal paste (that's all I had lol) between it and the boiler.

This is the thermocouple I ended up using: https://www.amazon.ca/dp/B00N2QTHLM/ref=pe_3034960_236394800_TE_dp_1

dcalacci, 2020-05-13:
first of all thanks for this project, super helpful and informative. 

for what it's worth... I just went through recreating it. not knowing what I was doing, I tried to use the type K thermocouple available from sparkfun, which is also ungrounded (and unshielded) : https://www.sparkfun.com/products/251 

I tried the hail-mary approach of securing it under the previous thermostat and... it worked. I absolutely don't think it's a permanent solution. Anyone done anything that they feel is really solid?

obiben, 2020-05-14:
I haven't gone through any serious steps to test this assumption and I've recently switched to a Rocket machine(so wont be testing it out), but my intuition is that the thermocouple works when you put current through the heating coil. 

My fix was to make the heating period longer at least a second and put a minimum 0.1 second heating period  to make sure the sensor picked up a reading and didn't ever stop recording after shooting passed the target. 10% heating time, even with a silicone-insulated boiler, wasn't over what heat the machine was losing so it worked fine for me.