brycesub/silvia-pi #19: Explanation

therealjohn, 2018-02-18:
Would you be able to explain this logic? 

https://github.com/brycesub/silvia-pi/blob/master/silvia-pi.py#L30-L44

What's unclear:
- Why a threshold of `100` for `avgpid`? 
- Significance of the `sleep` times for cycling heat on/off when `avgpid > 0 and < 100`

Thank you!

brycesub, 2018-02-18:
Each loop is 1 second long.  When the avgpid is >100 then that means the heating element should be on for 100% of the next second.  When it's between 0-100 the heating element should be on for that fraction of the next second, e.g. 70 means the heating element should be on for 7/10 of the next second.

therealjohn, 2018-02-19:
Thanks for the response? What’s the significance of 100?

brycesub, 2018-02-19:
That's how a PID algorithm works, you give it a goal, some inputs, and then it calculates an output that we use to drive the heating element.  It's a versatile algorithm used in other applications besides temperature control, for example the amount of power delivered to a motor to achieve a given RPM goal.

This link helped me the most when researching this project: http://m.eet.com/media/1112634/f-wescot.pdf

therealjohn, 2018-02-19:
@brycesub Thanks for the link. I'm OK with the PID concept. What I don't understand is exactly how it's used here. I don't see any clamping on the output of the PID or what the significance of 100 is in this context. Is it some arbitrary threshold of the PID output that you tested? Is it degrees? 

I've assumed [this](https://github.com/ivmech/ivPID/blob/master/PID.py) is the source for the `PID` being used. I see it clamps the integral term to -20/20 by default. But the output is not clamped. 

I've been playing with this and porting it to work with a Netduino but there are some differences I am experiencing. I noticed that if the temperature is above my set point, and hit this > 0 < 100 code, the sleep times are not sufficient to let the temperature cool down to the set point. Instead, it continues to heat slowly. 

Before I tweaked that, I wanted to understand it. Thanks for your help!

therealjohn, 2018-02-19:
I _think_ I've figured it out now. It's clearer now how to correctly time-proportion the PID output to something that can be used to determine the on/off signals to the SSR. I don't see where that's done here but I did manage to solve my issue. 

I'd still be curious if you could explain exactly how your PID output is normalized between 0-100. 

Thanks!