brycesub/silvia-pi #20: GPIO Pins

adamhratcliffe, 2018-08-12:
Hello,

I am using a Pi Zero W for my machine. I just wanted to check which pins to use.  In the high level diagram it suggests one layout then your images use a different layout.

Thanks very much in advance.

brycesub, 2018-08-13:
Use the layout in the pictures and you won't have to make any modifications to the config.py.  If you use the pinout in the wiring diagram you'll need to modify the config.py.  Either will work, but going by the pictures is probably easiest.

closed 2018-08-13 by brycesub

adamhratcliffe, 2018-08-14:
Thanks for your help. 

I tried both layouts. The led on my breakout board doesn’t seem to illuminate regardless of which layout I choose. 

Do I need to do any preceding steps before your guide to enable the GPIO pins?

Thanks very much in advance. 

Thanks,
Adam