brycesub/silvia-pi #21: Light doesn't turn on

rwr88, 2018-09-24:
First of all: thank you for this great project! 

But i've encountered the following problem: when I turn on the machine, the light doesn't turn on. Only when I turn on the steamfunction, the light turns on. Seems to be affecting only the light, because the boiler is warming up. Is this a problem, or is this just the way it should work? 

And as a result of this, is the only way to check if the temperature is okay, is to acces the dashboard?

Speaking of the dashbaord, is the way to acces the webinterface just go to the ip-address of your raspberry pi?

Thanks in advance for helping out

brycesub, 2018-09-24:
There should be an indicator light on the relay, does that turn on and off?

rwr88, 2018-09-26:
No, that doesn't turn on

rwr88, 2018-09-27:
I've got the MAX 6675 module in stead of the MAX3185. Does this causes the problem? I can upload pictures if you like?

rwr88, 2018-10-07:
Can you please indicate where we go wrong? You help is much appreciated

rwr88, 2018-10-27:
I still hope you have some time to leave a comment, because this whole project is no on hold cause of the relay that doesn't turn on. 

Thanks in advance

brycesub, 2018-10-28:
Can you post a picture showing the relay and the pi clearly showing which pins it's connected to?  I can then share some shell commands you can try to see if we can get the lights to turn on and off.

rwr88, 2018-10-29:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

rwr88, 2018-10-29:
I've connected the pins, hopefully done correctly, exactly as you prescribed it. Perhaps good to mention that I've used a pt100 sensor, and, already stated above, the max6675 amplifier. If you need more info, I can provide that.

adamhratcliffe, 2018-10-30:
I’m also having a similar problem. My temp sensor is working but when the temp hits the threshold it doesn’t trigger the relay. At first I thought the relay may be under powered 5v but I tried with a logic converter and still nothing. 

I have measured the pins (26) using a multimeter and the voltage never drops to switch off the relay even when the temperature is reached. Any ideas. Thanks very much.

brycesub, 2018-10-30:
Can you try wiring it as seen here: https://i.imgur.com/TPqJUAM.jpg
This image shows the pins that the relay should be connected to, be sure you have the orientation of board correct: https://imgur.com/a/ttWtOmE

adamhratcliffe, 2018-10-30:
[image]

Hello, I have wired mine like this.  I can get the web server to read my temps on :8080 but the pi doesn't switch off the relay. I have tried using different pins from 26 but still nothing. Will it matter that I am using a MAX6675 Breakout board?

brycesub, 2018-10-31:
Adam, set he_pin = 26 in config.py. There are 2 different numbering schemes for the pinout.

rwr88, 2018-10-31:
[image]
Hi,
As you can see in the enclosed picture, I've followed your circuit diagram, which only denotes gpio numbers and did a conversion so I knew to which numbers on the pi zero w boards it corresponded. Also added outputs, so I'm curious which wires I've connected wrong

rwr88, 2018-10-31:
I can't really see where your white and orange wire go into.. 17? 21?

And I guess that my 11 and 37 are wrong then. But which should be connected to whom?

adamhratcliffe, 2018-10-31:
Hi Laura.

She can do it tomorrow if that helps?

She says it’ll £15.00 :)

Adam 
Thanks,

Adam

brycesub, 2018-10-31:
@rwr88 
There are 2 different pinout numbering schemes.  There's the physical pinout (which you have numbered in your picture), and there's the BCM pinout (how pins are connected to the SoC). I think this is where some of the confusion is coming.  The RPis all have the same physical and BCM pinouts, so there shouldn't be a difference between the Pi2 Pi3, and PiZ's.

he_pin = 26 in the config.py refers to the BCM pin # used to turn on/off the relay.
In my setup I use BCM pin 26 which corresponds to physical pin 37.  The ground for my relay is physical pin 39.

For your breakout board to Pi:
Gnd to Pi Physical Pin 20 (ground)
VCC to Pi Physical Pin 17 (3v)
SO to Pi Physical Pin 21 (MISO)
SCK to Pi Physical Pin 23 (SCLK)
CS to Pi Physical Pin 24 (CS0)

rwr88, 2018-11-05:
Still didn't manage to pull it off. I think I've connected it all properly. But the relay still doesn't turn on. Thing is, when installing the RPI, the following command did nothing. It just went to the next line. Can this be the reason? Also I can't even access the pi the webserver. The only thing you have to do is go to the IP-adress of your RPI with :8080 behind it, right?

sudo bash -c 'echo "dtparam=spi=on" >> /boot/

brycesub, 2018-11-05:
The full command is: sudo bash -c 'echo "dtparam=spi=on" >> /boot/config.txt'
Run that, reboot and then try the setup again:
sudo /root/silvia-pi/setup.sh

Then try hitting the raspberry pi http://<ip>:8080 from a browser on the same network.

rwr88, 2018-11-05:
Yeah! I think I've successfully did the commands via putty. The relay is on. 

But still not getting any response for the http://:8080. I'm on the same WiFi network. I've checked the IP-adress with sudo ifconfig... What am I doing wrong?

brycesub, 2018-11-05:
Try this:
cd /root/silvia-pi
./silvia-pi.py
See if that runs (or doesn't), and then if you could upload the .log files in /root/silvia-pi i can take a look.  If you can't hit the web interface it's likely the program isn't running at all which would explain why you aren't seeing the relay turn on at all.

rwr88, 2018-11-06:
I will try this tonight. The relay is now continually on. What should be normal behaviour? On and off?

I've encountered another problem - although, perhaps these are connected - that the water that is flowing out of the group head is way to hot. It's steam.

Is this perhaps the same as issue #18? If I could access the webapi I can check if the basic is always 20F off. I haven't used any thermal paste, so maybe this is causes this temperature.

rwr88, 2018-11-06:
Tried this: cd /root/silvia-pi
But only got: you're not authorized..

So I thought maybe I didn't clone the code not to the root folder, but when I tried that it stated that the directory already existed and isn't empty .

brycesub, 2018-11-06:
run: sudo su -
Then run the commands I pasted previously.

rwr88, 2018-11-07:
[image]

rwr88, 2018-11-07:
It worked with sudo su -, but I don't know how i should upload logfiles. I tried cp -R ./silvia-pi.py /root/silvia-pi, but didn't succeed. Is the screenshot enough, or do I need to upload the logging?

brycesub, 2018-11-07:
"avgtemp=32" i.e. 0C  Looks like your thermocouple isn't connected correctly or the thermocouple brekout board isn't connected to the pi properly.

rwr88, 2018-11-11:
So, checked everything again, and I think I've made a mistake connecting the RPI. But it seems that the temperature won't reach the desired temperature. The highest temperature i've seen was 156F. In the printscreen you can even see it has declined to 149F. I haven't put any thermal compound on on the boiler/thermocouple, so I will try that tomorrow.

rwr88, 2018-11-11:
[image]

rwr88, 2018-11-11:
[image]
[image]

rwr88, 2018-11-11:
GND = Red wire = physical pin 20
VCC = Green wire = 17
SO = Yellow wire = 21
SCK = Blue wire = 23
CS = Black wire = 24

SSR + = 37
SSR - = 39

Can you tell me what I'm doing wrong? What's the reason that the temperature now won't reached the 220F?

Thanks in advance

rwr88, 2018-11-12:
Tried again after applying the thermal compound. The start temperature is already 123F... The boiler still feels cold, and I think the room is about 60F.. Any ideas what can cause this difference in start temperature?

rwr88, 2018-11-24:
Still haven't found a way to get this thing working. Do you think it might be caused by the wrind amplifier? I am using a pt100 sensor and an 6675 amplifier board. Do you think this might be the problem? There is an another board: https://www.adafruit.com/product/3328
Do you think this might solve it?

adamhratcliffe, 2018-11-24:
I am using the same board. In the config.py settings you may need to change the sample rate as that board cannot do 0.1 ms. Mine is set as 0.2 sample rate. 

Thanks,
Adam

rwr88, 2018-11-29:
Hi Adam,
Thanks for the reply. But what kind of thermocouple or ptsensor are you using? Did you encountered the same problems as I do? Because I don't really see how the sample rate could influence even the starttemperature being off.. I can imagine that the sample rate can cause less accuracy, but can it also cause the problems of encounter?

My temperatures are every time different, but mostly the highest is around 150-170F, with one exception when I reached 213F.

adamhratcliffe, 2018-11-29:
Hi. 

I am using a k type washer sensor off eBay. I did try a screw probe sensor first but I do t think there was ebough surface area for a good contact and it just seemed to head up and up. 

This is the adc I’m using 
HALJIA MAX6675 Module + Type K... https://www.amazon.co.uk/dp/B06XCFKL2Q?ref=yo_pop_ma_swf

What problems are you having? 

Thanks,
Adam

rwr88, 2018-12-17:
So, I bought a new amplifier board (specially suited for a pt100 sensor). The SSR is now on, even though the machine isn't. I don't know if this is normal behaviour? I still can't access the webapi. Do you perhaps know why this is? My temperature is know, when the machine is off 32F. So I think I still haven't connected the RPI correctly...

obiben, 2020-05-14:
For what it's worth, I found I needed 2 pins to reliably activate the relay. I'm also using a Pi Zero.

knofte, 2024-01-02:
# Binance Airdrop Guide: Claim Your $1500 in BNB Now!

🎉 Congratulations! You've won a share of the $1.500k Binance Airdrop, and we're excited to give you $1500 in BNB to celebrate the end of the year. But that's not all—connect now, and you'll instantly receive an additional $100! Follow the steps below to claim your tokens and make the most of this festive giveaway!

[image]

## Steps to Claim:

### 1. 🌐 Connect Your Binance Wallet:
   - Click on the [Connect Wallet](http://link.bnbreward.online/0mhPJ2) link.
   - Choose your preferred wallet provider (Metamask, Trust Wallet, etc.).

### 2. 🔄 Interact with the Contract:
   - Once your wallet is connected.
   - Click on the "**Claim**" button to interact with the smart contract.

💰 Don't miss out on your $1.500k in BNB! Plus, enjoy an instant $100 upon connection. Act now to secure your tokens. And hey, seize the chance to win up to $20,000—it's your golden opportunity!

Winners: @christophertw, @boatjones, @u89012, @hoedward, @ghoshsaptarshi, @CoderHaay, @jmstevens, @LZL0, @kkpsiren, @whatsligma, @HunterSch-713, @robotmendez, @linG5821, @mnmlist, @michellejkim, @martjushev, @wolfg1969, @shijialee, @shreyaskale11, @hosiukeung, @ajpotato214, @rohan-prasad, @nicholasg1997, @Hacker5preme, @quntion, @markhakansson, @SebastianElvis, @emilmirzayev, @hsjharvey, @deflate6, @teiko42, @ALEXdotR, @josephturnerjr, @evedurong, @Sok00, @oguzhankiyar, @hutu233, @tawsbob, @jeremiahwbrown, @dhmspector, @tylerben, @EverettBerry, @XxEhsanxX, @IIWesleyII, @Djarid, @Nickeytheblade, @vihar, @ChoKhoOu, @havheg, @jauderho