brycesub/silvia-pi #22: Pin 26 is always 3.3 Volts

Arkerr123, 2019-01-05:
First off - Thank you for this wonderful project.  I have been disappointed with my Gaggia Classic, and I am incredibly excited to get this running!

My current setup:
1: Dashboard works, and accurately represents temp of boiler.
2: SSR light is always on - The DC wires are connected to the GPIO pins 25 and 26.  My voltmeter confirms its always outputting 3.3VDC when your program is on.
3: The output of the program in terminal does show "Heating true when cold" and Heating False when over the setpoint.  

Do you have any suggestions as to what I can do to fix pin 26 from always being on?

Thanks,
Andrew

[image]

adamhratcliffe, 2019-01-06:
Do you have a picture of your GPIO pinout. 

Thanks,
Adam

Arkerr123, 2019-01-06:
[image]
[image]
[image]

Using a Raspberry 3 B+

adamhratcliffe, 2019-01-06:
Try moving your white ssr wire to physical pin 37. 

Thanks,
Adam

Arkerr123, 2019-01-06:
Works perfect now!  I did not understand the two difference between physical pin and GPIO Pin.  Thanks!

Off to taste testing now.  I will probably end up switching over to your newer version that includes a wake/sleep timer. 

Wohoo!

-Andrew

closed 2019-01-06 by Arkerr123