brycesub/silvia-pi #23: No values webinterface

rwr88, 2019-01-20:
[image]

So, as you can see above, I finally managed to get the webapi going. I bought a new amplifier (https://www.adafruit.com/product/3328), the MAX31855, because I've a PT100 sensor I thought it might has something to do with this.. This has the following outputs:
- GND
- VIN
- 3vs3
- CLK
- SDO
- SDI
- RDY
- CS

I've connected them in the following manner:
- VCC (green wire) to RPI physical pin #17 
- GND (red wire) to RPI physical pin #25 
- SO (yellow wire) to RPI physical pin #21 
- SCK (blue wire) to RPI physical pin #23
- CS (black wire) to RPI physical pin #24

SSR is connected to:
- SSR + (red wire) to RPI physical pin #37
- SSR - (white) to RPI physical pin #39

Also we've measured on the amplifier the values of the PT100 in ohm. This is functioning properly. 

So the question is: why aren't we receiving any values on the webapi? Is this because the amplifier isn't giving the signal to the RPI?

rwr88, 2019-01-20:
[image]

[image]

I've enclosed the electrical scheme i've drawed, based on the original scheme of Rancilio.

rwr88, 2019-01-20:
Images weren't visible, so i've enclosed them for a second time + some extra pictures of how wired it.
[image]
[image]
[image]
[image]
[image]

adamhratcliffe, 2019-01-20:
In the config.py check the sample rate. My sensor only works if the rate is
0.2 and above.

Adam.

rwr88, 2019-01-20:
Hi Adam,
Can you explain how i can check that?

adamhratcliffe, 2019-01-20:
Once you have SSH’ed into your pi use the following command  sudo nano /root/silvia-pi/config.py
Thanks,

Adam

adamhratcliffe, 2019-01-20:
> Adam,
> Thanks so much for your help. I feel like a complete noob: but how can I save the change?

Hello.

You press control and W

Thanks,

Adam

rwr88, 2019-01-20:
So, i've tried a bunch of stuff.. but when I exit and do another SSH session, the config is still 0.1..

adamhratcliffe, 2019-01-20:
Right so it definitely isn’t saving.

At the bottom the the nano window it should say which buttons to
Press to save. It should give you a prompt as you exit to save or not.

rwr88, 2019-01-20:
[image]
[image]
First screen is the home screen, second is when I press CTRL+W. There is an option that's called 'Opslaan' (Save), but that only says ^O... when I enter this, nothing happens

rwr88, 2019-01-20:
Also, the webinterface doesn't work from time to time.. It should be accesible all of the time right?

[image]

And the problem no is: the heating element is triggered, but it is never shut off. So it just keeps on heating, even if the desired temp is reached.

rwr88, 2019-01-22:
Adam, (or Bryce) 
Do you know how I can make adjustments to the config? Or better: how I can save those changes?

rwr88, 2019-01-22:
Ah, changing it to 37 might do the trick. So, am I running headless.. can you perhaps show me how it looks like if you're changing something I'm the config?

adamhratcliffe, 2019-01-22:
I can do but you'll have to wait for a few hours until I am home.  Have you
managed to find a way to save it control and x doesnt work?

rwr88, 2019-01-22:
Yeah, you're right..it was ctrl x.. now I've saved both changes

rwr88, 2019-01-22:
No, it did switch off.. but not on the right temperature. And so I did a check: I lowered the goal temperature to even 100F... But it didn't switch at the that temperature... So perhaps the switch was triggered by the overheat safety. So I thought perhaps the pi should reboot, so I unplugged it and did a restart.. but that blew it completely. I can't even connect to it headless right now...

rwr88, 2019-01-22:
So, did the whole install again. 

Conclusion:

Once again: it switched off, but not with the right temperature. When I fill in, in Putty, /cd/root/silvia-pi ./sivia-pi.py The values that are returned are: tempf: 32.... The webinterface also shows this.. 

What you said earlier about the config: Bryce said the following regarding this: "he_pin = 26 in the config.py refers to the BCM pin # used to turn on/off the relay. In my setup I use BCM pin 26 which corresponds to physical pin 37. The ground for my relay is physical pin 39" 

I've wired the SSR+ to 37, SSR- to 39. 37 is equivalent to BCM pin 26... so that should be correct
I've also GND (red wire) to physical pin 20 (GND). There shouldn't be a difference between which ground you use, right?

When I had the problem earlier, when the temperature stated 32F... Bryce said the following: 
 ""avgtemp=32" i.e. 0C Looks like your thermocouple isn't connected correctly or the thermocouple breakout board isn't connected to the pi properly."

But I just don't know where the fault lies..

rwr88, 2019-02-03:
@adamhratcliffe or @brycesub don't you guys have any clue what the problem is here? Could it be that the soldering of the amplifier of rpi is incorrect? 

Or have I bought the wrong amplifier? Does it only works with the bryce mentioned in the hardware section? Or have wired it wrong? See the pictures below..

[image]
[image]
[image]
[image]
[image]

obiben, 2019-05-18:
Hey rwr88, don't know if you ever found your solution. I've had problems with the pi 0 and max 31865
I've been using this class to read temps off the board:
https://github.com/obiben/silvia-pi/blob/master/max31865.py
It'll do all the bitbanging you need to do on any pins without using a SPI interface

The full story:
https://forums.adafruit.com/viewtopic.php?f=60&t=140006

I'll be receiving a MAX31856 any day now, since the 31865 has gone kaput and is intermittently oscillating between -256C and 900C. Will update my code shortly.

rwr88, 2019-06-29:
Hi @obiben i didn't found any solution. i even tried switching to the MAX31855, because i thought this might won't work with the PT100 1/3 din i've bought.

So, i would like to try again with the 31865 and the class your using to read temps. But how do achieve this?

Is it just to replace the following:

sudo git clone https://github.com/brycesub/silvia-pi.git /root/silvia-pi
sudo /root/silvia-pi/setup.sh

With:

sudo git clone https://github.com/obiben/silvia-pi.git /root/silvia-pi
sudo /root/silvia-pi/setup.sh

Because i'm not using a touchscreen display..

obiben, 2019-07-01:
Well you can start by just trying out the max31865.py class on its own. See if it prints temperatures for you.

You can get it from the link above and change the pins to your pins numbers in this section:
```
if __name__ == "__main__":
	try:
		csPin = 5
		misoPin = 19
		mosiPin = 20
		clkPin = 21
		max = max31865(csPin,misoPin,mosiPin,clkPin)
		while True:
			tempC = max.readTemp()
			print(tempC)
			time.sleep(0.1)
	except KeyboardInterrupt:
		pass
	GPIO.cleanup()
```

run the script using `sudo python max31865.py`. If it prints values you can tell your sensor and breakout board work.

rwr88, 2019-07-02:
Sorry, i'm not a developer of anykind, so I need a bit more info than this. 

So, I need to:
1) Install Raspbian
2) Make sure I can SSH to it
3) Connect the MAX3186 as previously to the RPI 
4) Fork https://github.com/obiben/silvia-pi/blob/master/max31865.py to my own
5) Change the pins to my own pin numbers, i.e.:"

VCC (green wire) to RPI physical pin #17
GND (red wire) to RPI physical pin #25
SO (yellow wire) to RPI physical pin #21
SCK (blue wire) to RPI physical pin #23
CS (black wire) to RPI physical pin #24

6) run the script using sudo python max31865.py ?? I don't think this is correct, so I probably don't understand.

Thank you very much for helping me. Really appreciating it!

rwr88, 2019-07-08:
Hi @obiben, could you please help me? See above for the questions

Thanks in advance!

obiben, 2019-07-08:
Your 6 steps were correct. If you want things to be simpler you could just set yourself up directly on your raspberry pi with a screen, keyboard and mouse until you're ready to put everything in a coffee machine.

Running the script in the command line is what I was telling you to do indeed. In a terminal, in the same folder as the file, if you type in 
`sudo python max31865.py`
You will execute that script. The script contains two parts: the class that I use in my own version, and the __main__ part that is executed when you run the script on its own.
Executing this script on its own will simply setup a max31865 object, read its temperature every 0.1 second and print it to the command line until you stop it (ctrl+c).

This wont make your coffee machine work but is a first step in verifying your sensor works, then you can see if you have to change that or if there's a problem somewhere else.

rwr88, 2019-07-09:
[image]

hi @obiben 
So i did the following:
1) install rasbian
2) make sure ssh was on
3) connect max31865 to rpi
4) install necessary things:

sudo apt-get -y update
sudo apt-get -y upgrade
sudo apt-get -y install rpi-update git build-essential python-dev python-smbus python-pip
sudo rpi-update
sudo bash -c 'echo "dtparam=spi=on" >> /boot/config.txt'
sudo reboot

5) fork the obiben project
6) Make adjustments to: https://github.com/rwr88/silvia-pi/blob/master/max31865.py 

CsPin = 24
miso =21
mosi = 17 ?
clk = 23

Like Brycesub, I don't use SDI so I was a bit confused which pin I should use as MOSI. The Max31865 has the following pins:
**VIN**
**GND**
3vo
**SCK**
**SDO**
SDI
**CS**
RDY

The ones in bold I use.

7) sudo git clone https://github.com/rwr88/silvia-pi.git /root/silvia-pi
sudo /root/silvia-pi/setup.sh

After this, I received the following error:  Could not find a version that satisfies the requirement Adafruit-GPIO==1.0.3 (from -r /root/silvia-p                                                                                                         i/requirements.txt (line 1)) (from versions: )
No matching distribution found for Adafruit-GPIO==1.0.3 (from -r /root/silvia-pi/requirements.txt (lin                                                                                                         e 1))

The thing I find odd is, that - if i understand this correctly - it should install the stuff in the folder 'silvia-pi'. But when i'm in the root folder, and do the 'ls' statement, i don't see this folder..

Do you know where it's gone wrong?
Thanks for helping out

obiben, 2019-07-10:
You're trying to install my fork of this repo which is wildly different from the one here. All I advised you to do was to run the **one** python script to verify your sensor works.

The error you are getting is from the install script. The specific command it fails at is the pip install -r requirements.txt where it installs python packages into your python folders. I think the adafruit library needs to be installed via github, and is just in my requirements file because I didn't maintain the installation script, as per the `*** Outdated ***` notice in my readme.

A google search of this error would serve you well. In fact if you want to do this to your coffee machine it's my opinion you need to be willing to google your ass off any time there is a problem.

As for the pins, I don't know what to tell you. I'm not a master of this breakout or the code that makes it chooch. 
My advice is exactly the same as last time: run the `max31865.py` script. It should work on its own without trying to install everything from my repo. You need python, the `RPi.GPIO` python package and the single `.py` file. Try pins out. 
Your decision not to use a pin arbitrarily strikes me as odd. Try it and if it works, that's wonderful, if it doesn't, maybe try with the pin connected. As far as I can tell SDI=MOSI and SDO=MISO, but I could be wrong. Try things.

Again, be prepared to google the errors you get and fix them on your own. No one knows these things before they learn them ;) There could be a hundred more things to do to make this work and I wont be holding your hand all the way, just wanted to give you the one tip! Good luck!