brycesub/silvia-pi #28: MAX31855 does not support grounded thermocouples

the-dbp, 2020-04-10:
on adafruits official website:

The MAX31855 does not support grounded thermocouples - if the sensor touches ground the chip will return an error

the project should be updated to avoid grounding the thermocouple. if you're not getting an error you're not grounding you machine which is illegal in europe and generally a bad idea, or you're very lucky and having the thermal paste somehow act as electrical insulator.

TiStef, 2020-04-27:
Can you please explain what you mean with the above comments? Sorry not an expert in electronics. Thanks

> the project should be updated to avoid grounding the thermocouple. if you're not getting an error you're not grounding you machine which is illegal in europe and generally a bad idea, or you're very lucky and having the thermal paste somehow act as electrical insulator.

the-dbp, 2020-04-27:
If you follow the project describtion you will screw the type k thermocouple on to the boiler.
The boiler is (or at least should be) grounded.

According to official documentation on the "MAX31855", the thermocouple amplifier refered to in this project this doesn't work. 

And indeed several issues have been raised with this, and closed by electrically isolating the type k thermocouple from the boiler, but on the project it's still described incorrectly without any electrical isolation.

TiStef, 2020-04-27:
Thanks a lot for our reply. now it is clear. And you think it might still work because the thermal paste used when installing the wire to the boiler is acting as insulator, right?
I see many other PID espresso projects using the MAX31855 so I am not really sure why it is claimed not to work with grounded attachment points

the-dbp, 2020-04-27:
It doesn't work if you ground the thermocouple, it's not a claim, that's simply how it is.

As mentioned there's two ways that the thermocouple might not hit ground:
1) thermal paste (extremely unreliable)
2) thermally condictive tape (the solution i think this project should be updated with)
Anyone getting it to work is probably using one of these methods

the-dbp, 2020-04-27:
#18 is the same issue but was closed without the project updating.