brycesub/silvia-pi #29: Know nothing...

bosuur, 2020-05-31:
Hi There,

Found your project and can unfortunately not get it to work.

I know nothing about programming, so I am sorry for these dumb questions:

1. When i run "setup.sh" i get this error in the end:
"jaraco.functools requires Python '>=3.6' but the running Python is 2.7.16"
.. can you help?

2. It's then installed and then what?... I am not able to access any web page to set brew time and more.... Can you help?

Again.... sorry for the stupid questions.

BR Bo (Denmark)

exze, 2020-06-06:
Hi 
I am to trying to get it to work.
1.  This worked for me. sudo pip install --upgrade jaraco.functools==2.0
2. Cant get the web page to load . Soo something isn’t working

Mikael (Sweden)

bosuur, 2020-06-08:
Hi Again;

This does not work for me. wonder why.

Still get:

"jaraco.functools requires Python '>=3.6' but the running Python is 2.7.16"

I would appreciate if someone can help :-)

Thanks a lot.

BR

Bo

bosuur, 2020-06-08:
Hi,

Fixed it with: sudo update-alternatives --install /usr/bin/python3 python3 /usr/bin/python3.7 2

But access to web page is  denied.

BR 
Bo

srjwebster, 2020-06-15:
I've just come across this project as I'm going to buy a Silvia very soon, and haven't got this software running, but I am a developer. If you can't get to the webpage this is hosted at, check permissions of the www folder and the files within. 

Also make sure you're trying on port 8080 which means accessing it on (ip-address-of-your-pi):8080

The default port is 80, 8080 is a common one people use for internal projects.

bosuur, 2020-06-15:
Hi Samuel. 

Thanks for your message. 

 I have tried to reach the webpage on port 8080 but no luck. 

I think it's not that simple. To me it looks like a python 3 vs. 2 issue.

I am not a developer, so I have unfortunately no idea how to check permissions. 

I look forward for you to get the SW up and running.... and would highly appreciate if you could update us on tips and tricks on how to get this to work😊. 

Have a good day!

Bo

srjwebster, 2020-06-15:
In a command line on the raspberry pi try `python --version`

It'll tell you which version you're running.

If you type `ls -al` you can see the permissions of all files and folders in the directory you are in.

Good luck!

mosch91-syn, 2020-06-24:
I also had the same issue getting the Webserver running. I know that I had to make some changes in the silvia-pi.py file to get it running under python 3.6 (it was programmed in 2.x). 

I found the problem in the "rest_server" function in the silvia-pi.py file. It was a Problem with the os-package version
The function looks for "index.html" File in the "www"-subfolder
Line 213:

basedir = os.path.dirname(__file__)

This is correct in Python 2.x, but doesn't work in Python 3.x
I changed the code to:

basedir = os.getcwd()

Now the function finds the "index.html"-file for the webserver and everything runs smoothely.
Maybe that helps!

Cheers

bosuur, 2020-06-25:
Dear mosch91-syn and srjwebster,

I am struggling with this. And it admit that I have very little knowledge about what I am doing.

Could you by any chance help me with a full set of instructions from start to finish?

I start using the Raspberry Pi imager and install a fresh image.

Then i configure the wifi and allow ssh in the gui.

when connecting with the user "pi" I can see that it starts by using python 2x.

what do I do from here?

I know its a lot to ask and for someone that knows it must be trivial to deal with someone like me, but I would highly appreciate if you took the time...

Thanks a lot.

Br

Bo

bosuur, 2020-06-25:
Hi again,

I managed to change to python 3x, but then I get all sorts of errors when running the silvia-pi.py.

i corrected the missing parentheses on the "print" commands but then there was an error with "urllib2".... I am lost here :-)

srjwebster, 2020-06-25:
The project is a little old and in need of some repair. If you're not
familiar enough with Python to correct it, I'd say this project is
simply not appropriate for your use. I'd recommend just subscribing to
updates in case a member of the community repairs it and publishes the
results.

bosuur, 2020-06-25:
Hi srjwebster,

I understand but reading mosch91-syn's message above I had the impression that everything runs smoothely by doing the suggested change.... Ill keep an eye on any updates.

Thanks.

Bo

mosch91-syn, 2020-06-29:
As srjwebster said, if you're not familiar with at least basic python, this project might be too much for you...

In my post above I pointed you towards a solution for the web browser not being displayed. Of course, this wasn't my only code change. I had a lot more problems going from python 2.x to 3.x. This meant a lot of googling and debugging. Still I wouldn't confident enough to open a pull request, because my code might be too specific, e.g. I use a different temperature sensor.

Regarding your problem I can only tell you, that I had the same problem with urllib because the syntax changes from 2.x to 3.x.
Maybe you should try to get everything running under python 2.x instead of 3.x...

igor-elbert, 2021-05-07:
Hi @bosuur , did you make any progress with this?

I'm about to order parts and attempt this install. Any advice on parts and install is appreciated. I'm a programmer and can help with migrating code from Python 2 to Python 3.

bosuur, 2021-05-07:
Hi Igor.Unfortunately not. No one seemed to be willing to help a novice like me😐.Good luck.Best regards BoSendt fra min Galaxy
-------- Oprindelig besked --------Fra: igor-elbert ***@***.***> Dato: 07.05.2021  21.43  (GMT+01:00) Til: brycesub/silvia-pi ***@***.***> Cc: bosuur ***@***.***>, Mention ***@***.***> Emne: Re: [brycesub/silvia-pi] Know nothing... (#29) 
Hi @bosuur , did you make any progress with this?
I'm about to order parts and attempt this install. Any advice on parts and install is appreciated. I'm a programmer and can help with migrating code from Python 2 to Python 3.

igor-elbert, 2021-05-31:
https://github.com/rwr88/silvia-pi - seems to be a clone with multiple changes including migration to Python3