brycesub/silvia-pi #34: Web interface

frequency17, 2021-02-03:
Maybe this is obvious, but I need to install apache for the web interface to work, right? I've followed the readme steps and I get nothing from localhost:8080. 

After installing Apache, I get:

```
root@raspberrypi:~/silvia-pi# ./silvia-pi.py
Starting Scheduler thread...
Starting PID thread...
Starting HE Control thread...
Starting REST Server thread...
Starting Watchdog...
./silvia-pi.py:67: RuntimeWarning: This channel is already in use, continuing anyway.  Use GPIO.setwarnings(False) to disable warnings.
  GPIO.setup(conf.he_pin, GPIO.OUT)
Bottle v0.13-dev server starting up (using CherootServer())...
Listening on http://0.0.0.0:8080/
Hit Ctrl-C to quit.
root@raspberrypi:~/silvia-pi# 
```
and then it exits back to the command line and still nothing on localhost:8080. Am I missing some steps?

RoccoMathijn, 2021-02-03:
Apache is not necessary. This software is using cheroot library which will be installed by the setup.sh script.

You're accessing localhost:8080 from the pi or from your laptop/pc? (just asking). Localhost is referring to the pi so if you're on your laptop/pc you should replace localhost with the ip address of your pi.

And can you maybe check the logs? There should be a file `silvia-pi.log` in your working directory.

frequency17, 2021-02-03:
I've tried on the pi itself with localhost and via another computer accessing the IP address of the pi. 

silvia-pi.log is:
`Starting Scheduler thread...
Starting PID thread...
Starting HE Control thread...
Starting REST Server thread...
Starting Watchdog...
`

I haven't connected any of the wiring. I'm just getting the software part up. Would that cause a silent failure?

RoccoMathijn, 2021-02-03:
That could be it but I'm unable to test it myself. There's a seperate error log for pid errors. Anything in there? I would guess an error in the pid thread if you didn't connect your sensor. 

You could also maybe add these lines just before the while loop on line 344 in silvia-pi.py. Guessing one of them returns false

```
print("p.isAlive()")
print("h.isAlive()")
print("r.isAlive()")
print("s.isAlive()")
```

frequency17, 2021-02-03:
Everything came back alive except for the PID. 

For subsequent readers, I the code should be:
`  print(p.is_alive())
  print(h.is_alive())
  print(r.is_alive())
  print(s.is_alive())
`

frequency17, 2021-02-03:
And actually having the PID not hooked up is why this was exiting. The while line is:
`while p.is_alive() and h.is_alive() and r.is_alive() and s.is_alive():`

When i removed the p.is_alive(), the page loaded. Thanks for the help troubleshooting!

closed 2021-02-07 by frequency17