brycesub/silvia-pi #35: in pid.err.log: sef._device.cshigh = False.  SystemError: error return without exception set.

performlabrit, 2021-05-01:
Hi folks,
I've just built this thing, and am running into some errors.

My PID.err.log is showing the error:

`pi@raspberrypi:/root/silvia-pi $ tail pid.err.log
Traceback (most recent call last):
  File "/usr/lib/python2.7/multiprocessing/process.py", line 267, in _bootstrap
    self.run()
  File "/usr/lib/python2.7/multiprocessing/process.py", line 114, in run
    self._target(*self._args, **self._kwargs)
  File "silvia-pi.py", line 116, in pid_loop
    sensor = MAX31855.MAX31855(spi=SPI.SpiDev(conf.spi_port, conf.spi_dev))
  File "build/bdist.linux-armv6l/egg/Adafruit_GPIO/SPI.py", line 46, in __init__
    self._device.cshigh = False
SystemError: error return without exception set`

Some digging suggests that the MAX31855 library has been deprecated:
https://github.com/adafruit/Adafruit_Python_MAX31855

...and I believe it is no longer compatible with SPI.py.    

I've already spent a few hours on this, so I'm going to punt for now.
Has anyone resolved this issue?

Is the easiest fix to install some older versions of the ...adafruit gpio libraries, maybe?

FWIW, this is on a pi-zero.

My thanks in advance!

RoccoMathijn, 2021-05-01:
Hey, your error seemed familiar. I googled it and behold ... all search results where purple.

Digged around and found I edited `/root/Adafruit_Python_GPIO/build/lib.linux-armv6l-2.7/Adafruit_GPIO/SPI.py` with the following edit and comment :
```
      46         # Apparently the below line is not necessary and it breaks so I try and comment it out
      47         #self._device.cshigh = False
```

Thank past me

performlabrit, 2021-05-02:
Thanks, past RoccoMathijn!  That did solve the PID issue.  No more errors, but still no webapi?!?  

No errors in the logs, either.  Hrummn....wish me luck :)